Xinyuan-LilyGO/LilyGo-AMOLED-Series #94: ARDUINO_RUNNING_CORE

kennygarreau, 2025-03-03:
Quick question on the preprocessor settings in T-Display-AMOLED.json

    "extra_flags": [
      "-DLILYGO_TDISPLAY_AMOLED_SERIES",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1"
    ],

I'm a little green around the ears here, but is there a reason with the T4 S3 that the running and event cores are set the same? Is this just an artifact from internal debugging? Thanks!
-Kenny

lewisxhe, 2025-03-04:
This is the default setting for most ESPs.

kennygarreau, 2025-03-04:
Thanks. I'm seeing some odd behavior when using nimBLE with these directives set. I put some logging into the BLE & WiFi callbacks as well as in setup() and loop(). Here's my testing:

No flags (removed RUNNING_CORE and EVENT_RUNNING_CORE):
WiFi: core 1
BLE: core 0 (set by nimBLE)
Setup/loop: core 1
BLE callback works

"-DARDUINO_RUNNING_CORE=1",
“-DARDUINO_EVENT_RUNNING_CORE=0"
WiFi: core 0
BLE: core 0 (still set by nimBLE)
Setup/loop: core 1
BLE callback works

"-DARDUINO_RUNNING_CORE=1",
“-DARDUINO_EVENT_RUNNING_CORE=1”
WiFi: core 1
BLE: core 0
Setup/loop: core 1
BLE callback **does not work**

I can certainly raise an issue on nimBLE's issues page, but just curious if you had any thoughts here. I would think we'd want the BLE and WiFi callbacks running on core 0 but perhaps I'm mistaken. Thank you!
-Kenny

lewisxhe, 2025-03-05:
Well, I can't help you.

sebus1234885, 2025-04-05:
> Quick question on the preprocessor settings in T-Display-AMOLED.json
>
> ```
> "extra_flags": [
>   "-DLILYGO_TDISPLAY_AMOLED_SERIES",
>   "-DARDUINO_USB_MODE=1",
>   "-DARDUINO_RUNNING_CORE=1", [...]