Xinyuan-LilyGO/LilyGo-AMOLED-Series #95: No display on my LILYGO S3 AMOLED Plus

Danou21, 2025-03-05:
I just received my LILYGO S AMOLED plus, I tried several examples like TFT_eSPI_Sprite.ino  and I have no display on the screen. I use ADUINO IDE

lewisxhe, 2025-03-06:
When you first received it, could it be turned on?

Danou21, 2025-03-07:
Yes it worked well

lewisxhe, 2025-03-08:
So it worked fine when you got the board, but stopped working when you uploaded a sketch? Is that right?

Did you select the correct board and configuration as described in the README?

What messages do you see when you open the serial?

Danou21, 2025-03-08:
It is right for your first question.
I already use the T-Display S3 AMOLED module (H717) without problem.
my configuration is
- ESP32S3 Dev Module V2.0.17
-LVGL 8.2.11
I downloaded the sketch "Arduino GFX HelloWorld" I get the serial messages:

"The sketch is only compatible with T-Display-AMOLED 1.91 inches, other sizes are not supported"
"CST816T init success!"

lewisxhe, 2025-03-10:
OK. Then you should know that this example is not suitable for your board. Try another example to test it. In platformio.ini, there are descriptions of which examples are suitable for special boards.

[image]

Danou21, 2025-03-10:
Sorry i'm not an expert.
 I am using arduino.ide, can you give me a simple sketch that displays a message on the screen,
 thanks in advance

lewisxhe, 2025-03-10:
There are lvgl and tft Sprite examples available. You can directly use this one https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/tree/master/examples/TFT_eSPI_Sprite for testing.

Danou21, 2025-03-10:
this sketch works , thanks a lot for your help and patience
best regards

closed 2025-04-24 by github-actions[bot]