Xinyuan-LilyGO/LilyGo-AMOLED-Series #97: Open a CD card in windows

Navsikaya, 2025-03-19:
Hello everyone

Is it possible to connect USB to LilygoT4S3 and open CD Card on Windows

LVGL version 8.4.0
Lily Go-AMOLED-Series version 1.2.0
ESP32S3 version 2.0.15

lewisxhe, 2025-03-20:
Please check the official guide
https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/jtag-debugging/index.html

Navsikaya, 2025-03-20:
@lewisxhe I read the information that I needed and did not find it. I use this code.
```c
#include` <LilyGo_AMOLED.h>
#include <USB.h>
#include <USBMSC.h>

LilyGo_Class amoled;
USBMSC msc;

bool mountSD = false;

// Функции для USB MSC (исправленные типы)
int32_t onRead(uint32_t lba, uint32_t offset, void* buffer, uint32_t bufsize) {
  (void)offset;
  return SD.readRAW((uint8_t*)buffer, lba) ? bufsize : -1;
}

int32_t onWrite(uint32_t lba, uint32_t offset, uint8_t* buffer, uint32_t bufsize) {
  (void)offset;
  return SD.writeRAW(buffer, lba) ? bufsize : -1;
}

static void usbEventCallback(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data) {
  if (event_base == ARDUINO_USB_EVENTS) {
    arduino_usb_event_data_t* data = (arduino_usb_event_data_t*)event_data;
    switch (event_id) {
      case ARDUINO_USB_STARTED_EVENT:
        Serial.println("USB PLUGGED");
        break;
      case ARDUINO_USB_STOPPED_EVENT:
        Serial.println("USB UNPLUGGED");
        break;
      case ARDUINO_USB_SUSPEND_EVENT:
        Serial.printf("USB SUSPENDED: remote_wakeup_en: %u\n", data->suspend.remote_wakeup_en);
        break;
      case ARDUINO_USB_RESUME_EVENT:
        Serial.println("USB RESUMED");
        break;

      default:
        break;
    }
  }
}

static bool onStartStop(uint8_t power_condition, bool start, bool load_eject) {
  Serial.printf("MSC START/STOP: power: %u, start: %u, eject: %u\n", power_condition, start, load_eject);
  return true;
}
void setup() {
  Serial.begin(115200);

 
  auto retry = 3; 
  amoled.begin();
 

  while (retry--) {
    mountSD = amoled.installSD();
    if (mountSD) {
      Serial.println("SD Card mounted successfully");
      break;
    }
    delay(500);
  }
  
  if (!mountSD) {
    Serial.println("SD Card mount FAILED");
    return;
  }
  
  USB.onEvent(usbEventCallback);
  msc.vendorID("LilyGo");
 
  msc.productID("T4S3-SD");

  msc.productRevision("1.0");

  msc.onStartStop(onStartStop);
  msc.onRead(onRead);
  
  msc.onWrite(onWrite);
  
  msc.mediaPresent(true);
  

  size_t sector_size = 512;
  size_t sector_count = SD.cardSize() / sector_size;//sd
  
  msc.begin(sector_count, sector_size);

  USB.begin();
 

  Serial.println("USB MSC Started");
}

void loop() {
  
}
``` 

It breaks down into "USB.begin();" and he writes that it is necessary to insert the D disk.

closed 2025-05-04 by github-actions[bot]