Xinyuan-LilyGO/LilyGo-AMOLED-Series #98: Charge display

Navsikaya, 2025-03-26:
Hello everyone 

Has anyone written an algorithm for the battery to display the percentages of the battery when charging. I have a certain algorithm, but turning on the USB power supply and immediately 100%, can someone tell me or is there a ready-made algorithm so that it would be like in a phone?

LVGL version 8.4.0
LilyGo-AMOLED-Series version 1.2.0
ESP32S3 version 2.0.15

lewisxhe, 2025-03-27:
You are talking about amoled plus.

It is impossible to calculate the percentage by reading the parameters, because the battery voltage will quickly rise to the full power voltage state when connected to USB, and the battery percentage cannot be determined by the voltage.

Navsikaya, 2025-03-27:
@lewisxhe  tell me, maybe there are some special libraries so that you can calculate or you need to think of an algorithm.

And what about amoled plus

lewisxhe, 2025-03-27:
Sorry, there is no such.

Navsikaya, 2025-03-27:
@lewisxhe Okay, thanks for your reply!

closed 2025-05-11 by github-actions[bot]