Xinyuan-LilyGO/LilyGo-AMOLED-Series #106: S3 amoled plus (or V2.0) display problem

sigma77, 2025-05-01:
Yesterday, my AMOLED display suddenly stopped functioning. Initially, I thought the display was dead, but after swapping it with another unit, it remains completely black. I tested the original display on that other unit, and it powers up, confirming that the display itself isn't damaged. 

I considered that the board might be dead, but my debug functions are printing correctly, so the ESP32 seems to be operating perfectly. 

I suspect that the LCD driver component may be damaged somewhere. I've checked for cold joints and signs of burn, but I haven't found any. 

Could you please advise on which components to check? I'd prefer not to return it under warranty if I can repair it myself! (Let’s save the planet...) 
A guide on this issue would also be beneficial for others, as I'm sure this is not the first and last time this will happen...

sigma77, 2025-05-01:
OK, got a bit closer:
went throught the displays datasheet, and did some basic research how these amoled displays work.
Also found the two testpoint on the board: EVDD,EVSS. (thumbs up for making it and labeling it! Thank you!)

On the perfectly functioning board i get 4.6V (EVDD) and -3.9V measurements. 

On the faulty board: 0V, 0V :(

Is the power managment ic dead?

lewisxhe, 2025-05-05:
Have you connected the screen for testing? The power IC must be connected to the screen for testing to have power output.

If there is still no output after connecting the screen for testing, then the AMOLED power supply is damaged.

sigma77, 2025-05-05:
Yes, it's connected. The power supply is functioning properly, but one of the capacitors near EVDD (possibly C35) was faulty. It showed a very low resistance to earth of about 90 ohms, whereas it should have been in the thousands or millions. What puzzles me is that both the schematic for S3-AMOLED and the OLED display datasheet indicate only one capacitor, which is 10uF (C35). In contrast, there are actually three capacitors present with significantly larger values. 

It might be wise to conduct a quality check on the production line, on these components anyway...

lewisxhe, 2025-05-05:
Can you take a picture of the fault? I need to investigate. Thanks.

sigma77, 2025-05-05:
[image]

All 3 of those caps inside the violet boundary is C35.
And i removed the one inside the red box on the left.

removed:

[image]

lewisxhe, 2025-05-05:
Thanks for the clarification

sigma77, 2025-05-05:
If I need to remove that cap from all of my boards because of a quality/design issue, I need to know.

lewisxhe, 2025-05-06:
What? What do you need?

closed 2025-06-19 by github-actions[bot]