Xinyuan-LilyGO/LilyGo-AMOLED-Series #109: Display Stopped Working After First Use - Only LEDs Light Up

anfelzirek, 2025-06-02:
After successfully displaying the default demo screen upon first connection, my LilyGO T-Display S3 AMOLED v1.0 now shows only a red CHRG_LED and green BOOT LED, with a completely black screen (no backlight or faint content), despite successful sketch uploads and attempts to reset brightness/flash factory firmware. The display stopped working after uploading a custom image sketch, suggesting either a firmware corruption or hardware failure during programming. Request guidance on recovery or replacement.

anfelzirek, 2025-07-03:
that means it's dead ?

lewisxhe, 2025-07-03:
The premise is that you write the correct firmware.
You can check whether the ESP32 starts by opening the serial monitor. If it starts normally but there is no display, it is dead.