Xinyuan-LilyGO/LilyGo-AMOLED-Series #111: Trouble with LilyGo-AMOLED-Series when upload file to esp32.

hmm-still-thinking, 2025-07-30:
Hi I encounter with this error "LilyGo_AMOLED.h:34:10: fatal error: driver/temp_sensor.h: No such file or directory
 #include <driver/temp_sensor.h>" I tried to downgrade esp32 core to 2.0.5 but the same. I'm using arduino IDE 2.3.6.

lewisxhe, 2025-07-31:
Update to 2.0.16 or latest. All examples are now tested based on 3.0.5.

lewisxhe, 2025-07-31:
All tests are passed in the latest version of the automatic component
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/actions/runs/16638430668

hmm-still-thinking, 2025-07-31:
Hi. I'd tried 3.0.5 to the latest but it still not woring. I'm using esp32 wroom 32E with a 1.47inch screen. Here is  my board setting. 

[image]

lewisxhe, 2025-07-31:
This library only supports LilyGo products

closed 2025-09-15 by github-actions[bot]